AxxAxx/AxxSolder #17: Stand temperature instead of full cooldown

blackfield3k, 2023-10-19:
First of all, thank you for this awesome project and the support you provided via email. First station is done and working!

I noticed that the iron fully cools down in the stand. As i remember the JBC stations just reduce the temperature to 150°C or something. Someone told me that would reduce the stress and oxidation of the tips. I can´t confirm that but it sounds plausible.

After 10-15 minutes the JBCs shut off the heaters completly.

AxxAxx, 2023-10-20:
Nice! Thanks! You are the first one I know to have built one, nice with some user-input! 

Yes! 
JBC calls this  "Sleep & Hibernation" and write: 

> Thanks to automatic detection of the tool in the stand, JBC Soldering and Rework Stations allow the tools to enter into **Sleep & Hibernation** mode when not being used. As a result, tip life lasts up to 5 times longer. 
>
>**Longer Tip life**
>Tip life increases exponentially by using lower temperatures as shown. Using the Sleep Mode, the temperature is further reduced, which extends tip life by 5.
>
>**Sleep Mode**
>Sleep Mode automatically lowers tip temperature below solder melting point when the tool rests in the stand.
It prevents the dissolution of the tip’s iron coating into molten solder.
>
>**Hibernation Mode**
>After a configurable period of tool inactivity in the stand (10 min factory default), the tool enters into Hibernation Mode.
It cuts off the power supply making the tip reach room temperature thus preventing oxidation and saving energy.

A graph can be seen here: https://www.jbctools.com/top10.html

I never really thought about shutting off completely as an issue, I underatand that elevated temperatures for a long time reduced the lifetime. 
Thermal cycling and fatigue is also a thing of course...
But okay, I assume JBS has put a lot of thought into this.

It should be fairly staright forward to implement.

AxxAxx, 2023-10-20:
I added a few lines to the code.
Lets call the ´new mode "STANDBY" 
After the iron is placed into the stand the temperature is set to STANDBY_TEMPERATURE (default 150 deg C) for X amount of time (default 10 min). After that the iron goes into sleep mode and switched off heating completely.

I will test the new implementation a bit more before I push it into the main branch. 
If you want to test you can find the BINARY here: 
https://github.com/AxxAxx/AxxSolder/tree/17-stand-temperature-instead-of-full-cooldown/Software/AxxSolder/Debug

AxxAxx, 2023-10-20:
Dont know if it is a feature or not but now you can get out from STANDBY to SLEEP with a buttonpress:
https://github.com/AxxAxx/AxxSolder/commit/6ba68a1f9d676d600e40142af75eae9afbca315f

You find the new AxxSolder.bin in the same place as before.

Good luck! :)

closed 2023-10-22 by AxxAxx