AxxAxx/AxxSolder #24: Rework station enclosure design

AxxAxx, 2023-10-25:
Improvement of station enclosure to have a proper closed enclosure, good locations for screw threads where needed.

f616, 2023-11-12:
Could you provide us with CAD files than can be edited in Fusion360?

f616, 2023-11-12:
This is in what I'm working now.

[image]

Wasn't yet printed and tested. If anyone finds value on this, and assuming the tests will be OK, please let me know and I can share the Fusion360 files.

AxxAxx, 2023-11-18:
> Could you provide us with CAD files than can be edited in Fusion360?

The CAD designs was all made in SolidWorks. Therefore I do not have any Fusion compatible design files. The best I have for you to build on is .STEP, which are availible at  https://github.com/AxxAxx/AxxSolder/tree/main/CAD

bennymeg, 2024-02-01:
It's also possible to import SolidWorks assemblies and parts directly into Fusion 360 now.

AxxAxx, 2024-03-01:
fixed in AxxSolder 3.0

closed 2024-03-01 by AxxAxx