AxxAxx/AxxSolder #25: LCSC.com Part Numbers for v2.1

f616, 2023-11-03:
Hi,

Those are the part numbers for v2.1 from [LCSC](https://lcsc.com)

| Part                                                      | LCSC.com Part Number |
|-----------------------------------------------------------|----------------------|
| C1, C6, C12, C14, C19                                     | C15850               |
| C2, C3, C5, C7, C8, C9, C10, C11, C13, C20, C21, C22, C24 | C49678               |
| C4, C17, C18                                              | C12891               |
| C25                                                       | C45783               |
| C15, C16                                                  | C46653               |
| C23                                                       | C310841              |
| C26                                                       | C113832              |
| D1, D3, D5, D6, D12                                       | C916425              |
| D2                                                        | C2687553             |
| D4                                                        |                      |
| D7                                                        | C503376              |
| D8                                                        | C707996              |
| F1                                                        | C352820              |
| IC1                                                       | C688540              |
| IC2                                                       | C529357              |
| L1                                                        | C2847543             |
| IC3                                                       | C2869735             |
| IC4                                                       | C882727              |
| J1                                                        |                      |
| J4                                                        |                      |
| LS1                                                       | C255328              |
| PS1                                                       | C115288              |
| IC6                                                       | C311983              |
| Q2                                                        | C5224254             |
| R1, R5, R6, R17, R25, R26                                 | C17673               |
| R2                                                        | C17713               |
| R3                                                        | C17772               |
| R8, R9, R11, R12, R13, R14, R16, R18                      | C17414               |
| R10                                                       | C346458              |
| R15                                                       | C17557               |
| R19, R20                                                  | C4310                |
| R21, R22                                                  | C17709               |
| R23, R24                                                  | C17556               |
| R27                                                       | C17720               |
| R28                                                       | C149504              |
| R29                                                       | C17446               |
| U1                                                        |                      |
| U2                                                        | C2760081             |

I've orders the boards, but please bear in mind I've not tested them yet.

AxxAxx, 2023-11-18:
Thanks! 
This could be useful for people using LCSC.com/JLCPCB !

Andy79881, 2024-02-15:
@f616 Thanks for your work regarding LCSC part numbers. I really appreciate that. Could you check for update to the latest HW release (V2.2)?

f616, 2024-02-17:
@Andy79881 updated the table with the components from v2.2
I've a few pre-popuçated boards v2.1, If you need one, please let me know

Andy79881, 2024-02-18:
@f616 Thanks for the update!

closed 2024-03-01 by AxxAxx