AxxAxx/AxxSolder #34: Individual tip calibrations

wildstar87, 2023-12-18:
One of the things that is great about the custom firmware for the STM32 T12/C245 stations, is the ability to save individual tip calibrations, since even with genuine JBC tips, different versions/models have varying temperatures, with the clone tips even more so.  I have both Unisolder and Aixun T3A, neither offers this capability, so if you have cartridges that don't match the built-in tip profiles, you are out of luck.  With Unisolder these profiles can only be modified through firmware change, if you understand the code, and how the author coded it.  With the Aixun T3A, you can change the calibration profile through a number of temperature steps, but it has two problems.  First, it is not tip specific, Two even though it saves the value, this is change is never displayed to the user, so when you go back into the menu, all setting show with no changes.

AxxAxx, 2023-12-19:
Good input. 
How would you name each of the tips? with their JBC number? 

Or would it make sense to have one "default" configuration and then the user can add "custom conf 1", "custom conf 2" and so on. 

And then maybe you can select the default start-up configuration in a settings menue?

wildstar87, 2023-12-19:
Given the number of variations in JBC tip numbers, putting in a pre-populated list might be troublesome, in addition to the number not being immediately descriptive, but I don't know how much memory space that would take up, if you put in all the possible tip model numbers.  The other thing is the clone tips often use a different numbering scheme that they came up with.  Unfortunately I've only see pictures on the STM32 T12/C245 stations, and I think that they are pre-populated tip numbers for the T12 at least, not sure if they have all of them, or just the popular ones.  Maybe having a limited character field, like 8-10 characters, where you can specify the name, using the encoder to scroll letter/numbers.

The custom firmware made by DavidAlfa, I think does have a base/default tip profile, then you can base other tips off that profile to start, but each tip has it's own parameters that can be modified, that can be done through just temperature calibration points.  

Not sure how many entries you would need.  Maybe look at his code to see what he's doing?

AxxAxx, 2024-10-08:
I will look into this in coming fw updates. Thanks for your ideas!

closed 2024-10-08 by AxxAxx