AxxAxx/AxxSolder #38: Maybe u can change CMI-9653S-SMT-TR for something more popular and available? Ordering it from mouser doesnt make sensie for me and i cant find it anywhere else.

AxxAxx, 2024-02-21:
Maybe u can change CMI-9653S-SMT-TR for something more popular and available? Ordering it from mouser doesnt make sensie for me and i cant find it anywhere else.

_Originally posted by @Froglol in https://github.com/AxxAxx/AxxSolder/issues/26#issuecomment-1956450665_

AxxAxx, 2024-02-21:
DigiKey, Mouser, RS Components all have these in very large quentities. 
https://www.digikey.se/en/products/detail/cui-devices/CMI-9653S-SMT-TR/6012408
https://eu.mouser.com/ProductDetail/CUI-Devices/CMI-9653S-SMT-TR?qs=qCxwlXJ4fnz7UpLQu8UKbg%3D%3D
https://se.rs-online.com/web/p/piezo-buzzers/2596214

I would say this is a quite common buzzer. 

But do you have any replacement part in mind? If you can find one with the same size and resonance freq. I can add that as a "Alternative Part" to the schematic. 

If not, this is not a critical part and yhou could easyly replace it with a buzzer with similar resonance freq and similar footprint. Doesn't have to be exact.

Froglol, 2024-02-21:
I just found something similar on alliexpress. Buying from Digi RS or Mouser doesnt make sense for me because the shipping costs as much as 10 od these. Ordering all parts from them also doesnt make sense also because on alliexpress u can get them in better prices(especialy resistors and capacitors when buying a sample book). For nów I will wait for 3 revision. Thanks for your work!

closed 2024-02-24 by AxxAxx

f616, 2024-03-01:
I've used a compatible from lcsc.com. please check issue #25