AxxAxx/AxxSolder #49: Automatically detect USB-PD supply capability and limit power

AxxAxx, 2024-04-04:
Automatically ask USB-PD supply of it's power capability with the STUSB4500 negotiation chip and limit the output power based on the capability. 

There is currently no such check and if you power AxxSolder with a 65W capable USB-PD source and try to use a C245 cartridge (which will draw 120W) AxxSolder turns off as the supply voltage can not be kept by the supply.

c0d3z3r0, 2024-04-26:
AP33772S might be a good replacement for STUSB4500 to add EPR support (240W). It requires a few more external components, though.
https://www.diodes.com/part/view/AP33772S?BackID=10099

AxxAxx, 2024-04-27:
Nice, they are finally arriving, the PD3.1 sink controllers! 
Seems like one has to wait a bit more to get the hands on the AP33772S IC's though. Not availible anywhere for now what I can see? 

I assume more IC manufacturers will release PD3.1 sink controllers too in the near future.

c0d3z3r0, 2024-04-27:
Oh oops, didn't check availability and just assumed it is :D There one ftom Hynetek as well, but also not available anywhere.

Yeah, maybe even G4931 with 512KB. Shouldn't need too much code change.

Hmmm, weird. Datasheet says PD spec 3.0 only o.O UCPD page lists G4 series

AxxAxx, 2024-04-27:
OK, haven't digged into it too much but [here](https://www.st.com/content/st_com/en/ecosystems/stm32-usb-c.html#stm32typec-why-use-usb-c) it says: 

STM32 UCPD MCUs are compliant with the latest USB PD r3.1 specifications and support:
- SPR messages (standard power range up to 100 W)
- EPR messages (extended power range up to 240 W)
- PPS (programming power supply)
- AM (alternate mode)

And lists STM32G4 series to have one UCPD interface 
Don't know.. :)

c0d3z3r0, 2024-04-27:
Yeah, maybe it only depends on the sw stack or the datasheet is outdated, no idea

c0d3z3r0, 2024-04-27:
At home again, so I could have another look....
- The UCPD marketing site says what you cited already (EPR, VBUS 48V, 240W) https://www.st.com/content/st_com/en/ecosystems/stm32-usb-c.html
- STM32G4* are named explicitly but datasheets mention PD 3.0 only
- Their UCPD wiki does only mention PD up to 100W, no EPR at all https://wiki.st.com/stm32mcu/index.php?title=Introduction_to_USB_Power_Delivery_with_STM32
- The protection chips TCPP* only support VBUS up to 22V and PD 3.0, max. 100W
- CubeMonUCPD only supports PD 3.0 https://www.st.com/en/development-tools/stm32cubemonucpd.html
- *X-CUBE-TCPP does support PD 3.1 EPR** https://github.com/STMicroelectronics/x-cube-tcpp

W....T....F....

AxxAxx, 2024-04-28:
Thanks for summarizing your finds!
Indeed the information is a bit all over the place. 

Maybe it will clear out with time as PD3.1 EPR is getting more mature

NightHawk32, 2024-05-27:
Power negotiation and dynamic power limiting is handled in #62

closed 2024-05-27 by AxxAxx

AxxAxx, 2026-01-09:
This is implemented in the firmware now. 
If the usb power supply would limit the current you do get into brown-out scenarios which is difficult to handle