AxxAxx/AxxSolder #67: Is support for tweezers possible, such as the AN115?

F10F4F9, 2024-07-15:
As mentioned in the title, this would be a standout feature that many are probably waiting for.

Thanks
Marco

AxxAxx, 2024-07-17:
It COULD for sure be added. But that is to an increased BOM cost. What would people be willing to pay extra for tweezer support? I estimate it to increase the BOM cost with 20 Euro or so and some added "complexity" for the user when connecting and sensing.

I think it it still better to have tweezer support in the AxxSolder and not create a separate hardware only for tweezers. Do you agree?

Everything is there in terms of software and hardware (for one heating circuit). It is a bit of work to implement in a new hardware in the future but not too much. What I have to look into is how the user would connect either tweezers, iron or dual iron (now when we would have two heating circuits) and how the tip sensing would work. Haven't figured that out yet.

The JBC tweezer is also designed to be heated with AC current. There are two diodes inside allowing one tip to heat up during the positive phase of the AC and the other during the negative phase. Either one removes the diodes and modify the tweezers to be heated from two DC sources or implement a H-bridge drive...

F10F4F9, 2024-07-22:
Yes, I agree with you; it should be integrated into the existing hardware and not create additional hardware just for the tweezers. Personally, I would avoid modifying the tweezers, even if it's a minor change. I believe compatibility should be maintained.

If you are looking for a tester and feedback, I could support that.

AxxAxx, 2024-07-24:
Sure! I will look into this and the different possibilities. It seems now that to drive the JBC tweezers you need a controllable AC current which could be achieved with a H-bridge drive. But thats a totally different topology than used now. I'll look more into it and see how it can be done and if it can give some other benefits.

I'll close this issue for now.

closed 2024-07-24 by AxxAxx