AxxAxx/AxxSolder #69: DFU Fails

BB-BenBridges, 2024-07-17:
Hi there

I'm having an issue updating firmware over DFU. I connect the device to my Mac whilst holding the right most button. CubeProgrammer will detect the device for 5-6 seconds and then nothing. This obviously does not leave enough time to upload the firmware. I do not have an STLink programmer to test with.

Any suggestions? I assume I don't have to keep the button held?

AxxAxx, 2024-07-17:
Hi!

No you do not need to hold down the button all the time. Only during the boot sequence.

Can you try
- a different cable
- Power up the AxxSolder from a DC source while pressing the button, then connect USB to your Mac. Do this without any handle connected.

//
Axel Johansson

BB-BenBridges, 2024-07-17:
Same problem! Tried 3 different cables, and multiple ports on my Mac. CubeProgrammer detects it for a few seconds and then it's gone again. I'll try a Windows machine tonight incase it's a Mac issue

AxxAxx, 2024-07-24:
Did you find any solution to this? 

It is difficult to help you with this as I can not replicate this problem on my machine..

closed 2024-08-10 by AxxAxx

AxxAxx, 2024-08-17:
Hi! 
I know what the proeblem is now! 
There is a bug in the newest ST programmer software (2.17):
Download version 2.15 instead and it should not throw the error. 
The Readme is updated:
https://github.com/AxxAxx/AxxSolder?tab=readme-ov-file#firmware-update

BB-BenBridges, 2024-08-19:
@AxxAxx Amazing! Thanks for heads up. Update worked perfectly now.

BB-BenBridges, 2025-06-05:
Correct, the latest version (2.19) worked fine. You can remove the instruction to use version 2.15