AxxAxx/AxxSolder #70: Hirose RPC1-12RB-6P(71) Obsolete

shakey2k, 2024-07-24:
Hi.  I stumbled upon your this and wanted to build a AxxSolder Portable.  I was in the process of ordering Hirose RPC1-12RB-6P(71) from mouser and it's now listed as obsolete.  I also tried some other electronics stores like digikey and they are also listing it as obsolete.  Do you know of an alternative for this connector end?

AxxAxx, 2024-07-24:
Hi!
Yes I know it can be difficult to source. But RS components have it as well as Farnell. I also have a bunch of them if you want to buy from me. 

But as fas as other alternatives I have not found any. JBC use them and I am curious to know how they will handle it if the connectors are really going to get retired in the future and impossible to source. 

Tell me if you find alternatives! :)

shakey2k, 2024-07-24:
I will consider buying one or two from you when your populated AxxSolder
PCB is available for sale.  When will it be available?   Depending on when
your populated AxxSolder PCB will be available, I might consider buying
your unpopulated AxxSolder PCB and do the soldering myself.  If I go that
route how do I purchase a Hirose RPC1-12RB-6P(71) from you also?

AxxAxx, 2024-07-24:
The populated boards will be available again when Im back from vacation in one week. Then it will be the AxxSolder 3.1 hardware. (There are no big difference to the 3.0 version) Just better hardware RC filter on the buttons and encoder. There is no difference in performance. Just done in a nicer "proper" way.

So I'll update the Tindie store in one week with this update. 

To buy just the connector I would suggest that you send me an email and pay me with PayPal. That will also be earliest in one week when Im back.

shakey2k, 2024-07-24:
How much for one connector and shipped to the US (zip code: 29445)?

AxxAxx, 2024-07-24:
Its 14 Euro for shipping to the US and connector

shakey2k, 2024-07-24:
Awesome.  Please send me a Paypal invoice ***@***.***) for two and
I'll pay.

I'll wait for you to have the AxxSolder version 3.1 populated board in
stock and will purchase that once your back from your vacation.

One quick side note.  I'm going to build an AxxSolder portable to use with
the JBC NT115 Nano Handle.  The portable case has an area for an aluminum
insert so that you can rest the collar of a JBC soldering iron on it to
activate sleep mode, but for the Nando Handle, the metallic portion is not
part of the collar and more in the rear of the handle.  How do you envision
using placing the Nano Handle on the AxxSolder portable to place it into
sleep?

AxxAxx, 2024-07-24:
GitHub removes the email adress. Email me. 

You have my email here: https://github.com/AxxAxx

AxxAxx, 2024-07-25:
And regarding the nano handle and sleep:
I usually rest the cartridge on to the aluminium plate. Not the collar necessarily. I believe this will work with the NT115 handle too.

closed 2024-08-10 by AxxAxx