AxxAxx/AxxSolder #77: Diode leakage current

electroniceel, 2024-10-04:
Your project looks interesting to me, especially the mobile version. Thanks for working on it and publishing it.

Before building my own I did take a look at the schematics. There was one thing that I noticed that could be an issue: diode leakage.

- D7 is used as protection diode for VBUS_IN. It is a BAT54S. I have used this diode type as protection for ADCs before too, and noticed considerable errors due to reverse leakage current of these diodes. Since then I switched to BAS70-04 which offer similar protection, but have much lower leakage.

- There are D5, D6, D8 that are also BAT54S on an analog input. In these places a bit of offset isn't that critical. But it would probably make sense to also switch them if you introduce one BAS70 to not increase the number of different BOM items.

- D10, between Blue and Red. It is a PMEG6020ER. I guess this is also used as protection diode, but I'm not 100% sure of it's function. For the C115 and C210 cartridges it looks to me like this diode is in the path of the thermocouple and leakage through this diode would create an offset in the tc voltage. To suggest a better alternative I'd have to be sure I understand all of it's functions.

Because diode leakage varies very much with temperature, these leakages are hard to calibrate out and make the device behave differently with temperature.

Please note: this issue is solely based on looking at the schematics. I have not done any measurements on an actual device. So it could very well be that this is no problem and the leakage doesn't pose an issue in practice.

AxxAxx, 2024-10-04:
Thanks! 
Good points, all three of them. Here are my thoughts: 

I understand your concerns about the BAT54S. In the early prototypes almost a year ago I started of with BAT54S at all locations as protection diodes. This is probably what JBC does in their stations. This is not confirmed by me but according to the great reverse engineering by johnmx at EEVBlogForum it is like this: https://www.eevblog.com/forum/testgear/jbc-soldering-station-cd-2bc-complete-schematic-analysis/

However, I know the BAT54S has problems with reverse leakage current and therefore these were changed to the BAV199  which has a very low typical leakage current of 3 pA. There are now two BAV199 at the thermocouple op-amps. The inputs for handle/stand sense signals and input voltage are not sensitive to this type of leakage current and it does not really matter - thus the BAT54S stayed. One could maybe change these to BAV199 to reduce the BOM lenght...

The idea with D10 has indeed been protection from inductive loads. Now I am not sure if it is strictly needed. And the leakage current of this could have an impact yes. I do not know exacltly how much. I have never seen any temperature drift of the AxxSolder from "cold start" at ~24 deg C to maybe a few degrees hotter after  using it a low in a warm day. I do not think it makes a difference in practise. But I see your point on paper. Maybe I'll look into this more in a coming version.

electroniceel, 2024-10-04:
Thanks for looking at this.

I have seen the BAV199 on the opamps. The Opamp inputs are usually specced with 10 mA current through the protection diodes, so the higher forward voltage of the BAV199 is fine there. So the BAV199 are an excellent choice for this location.

But the STM32 inputs don't allow 10 mA, there Vss - 0.3V is the lower limit. So a BAV199 would be a bit of a questionable choice there. The BAS70 is a bit of a middle ground there with more leakage than the BAV199, but it is still a Schottky with lower forward voltage.

Regarding D10 I was a bit irritated by the diode current capability. I wouldn't expect a constant current anywhere near 2 A there unless you have a serious inductor in the path. So maybe a BAV199 would work there too? It allows 160 mA continuous and 500 mA repetitive peak.

electroniceel, 2024-10-05:
If you think the BAV199 is a bit too flimsy to replace D10, maybe the SMF4L24AT1G you already use would be another good choice for lower leakage. In case of overvoltage due to inductance it would be used forward, which is a bit unusual for a TVS, but I don't see an issue with that in this case. It is specced at 100 nA max leakage at 24V, but when you sense the thermocouple you'd have much lower voltage, so even lower leakage. This is about 2 orders of magnitude better than the PMEG6020ER.

AxxAxx, 2024-10-05:
That is a great suggestion. I think for this design it is no major issue. But why not look into and implement this improvement in the next hw revision. 
I am anyway going to look into how to do "open thermocouple detection" (https://www.ni.com/en/support/documentation/supplemental/10/overview-of-open-thermocouple-detection.html) in a better way to detect when there is no cartridge in the handle. This is now done by looking if there is a sudden impossible change in temperature and at the same time the current through the heater can be probed and if there is no current flowing the cartridge is faulty OR removed. 

I will experiment with adding a 10 Mohm resistor between RED and 3.3V. this will drive the op-amp high and can be better detected by the MCU. Now the input "float" (D10 and a 5 mohm shunt is between RED and GND) when there is no cartridge detected. 

By doing this I will have to re-calibrate all cartridges so perhaps changing the diode at the same time could be good.

electroniceel, 2024-10-05:
> I will experiment with adding a 10 Mohm resistor between RED and 3.3V. this will drive the op-amp high and can be better detected by the MCU. Now the input "float" (D10 and a 5 mohm shunt is between RED and GND) when there is no cartridge detected.

Good idea for detecting thermocouple problems. Maybe make the 3V3 through the 10M resistor switchable by hooking it to a GPIO pin that you could switch between input (=floating) and push/pull high? So you could toggle it and compare the results with/without. Making it switchable would remove any influence on the TC measurement.

AxxAxx, 2024-10-05:
Thats a great idea. I'll go on vacation now a couple of weeks but I'll look into this for the next hardware release. Thanks for the input and ideas!

AxxAxx, 2024-10-08:
I will close this issue and bring it to my to-do list for upcoming hardware revision!

closed 2024-10-08 by AxxAxx