AxxAxx/AxxSolder #104: common problems

mdbetancourt, 2025-02-28:
Is common to station which uses the C245 cartridge to experience problems like voltage/current leakeage and error when circuit is grounded like [https://www.youtube.com/watch?v=B7jLpHgR7fI](aixun T3A) does this solder has any of these problems?

AxxAxx, 2025-02-28:
Hi! 
good question. 
Tony himself (who did the video in your link) asked this recently on the Discord server. 
https://discord.com/channels/1177970799697727548/1177970799697727551/1344310961024143401
You can read there so i do not have to write everything again. 

But to summarize: 
As long as you use a SELV (Floating output) power supply and provide a connection between mains protective earth (PE) and AxxSolder there is a galvanic connection between PE and the tip. By doing this there is no leak voltage on the tip. 

If you power AxxSolder from USB PD with no grounding there will be leak voltage on the tip

closed 2025-03-10 by AxxAxx

AxxAxx, 2025-11-14:
Indeed. You need to use a floating power supply in order to ground the tip. 

That is why this section is in the GitHub documentation:

https://github.com/AxxAxx/AxxSolder?tab=readme-ov-file#recommended-power-supply-for-tip-grounding