AxxAxx/AxxSolder #105: Why is used STUSB4500

mdbetancourt, 2025-02-28:
From the documentation the STM32G431CB supports usb PD 3.0 same as STUSB4500 there is any reason to use that chip?

AxxAxx, 2025-03-01:
Yes it does, 
two reasons it uses a external USB PD att the moment: 
1. The USB PD software does not fit the MCU at the moment, it is wuite big in size. 
2. I have used the STUSB4500 a lot in the past and had code for it already.

With that said, I would like to go away from the STUSB4500 and use the internal in the future, but that requires changing the MCU

closed 2025-03-10 by AxxAxx

mede-erik, 2025-03-12:
To solve point number 1 I would recommend switching to a STM32G491CE/STM32G491CC, this could allow you to remove the STUSB4500 and gain space on the board

AxxAxx, 2025-03-12:
Yes, for the next version I plan to use the STM32G491CCU6