AxxAxx/AxxSolder #111: problem with firmware 3.4

onatskiy-sergey, 2025-03-25:
There is no link to download the latest firmware 3.4 from March 19th.

AxxAxx, 2025-03-25:
Hi! 
The 3.4 is still in beta testing. 
These are some of the new features in 3.4:
 
- Including a setting to turn on a short beep when the set temperature is reached @Marin
- Enter the settings menu from long-pressing the encoder button
- Update the preset temperature by long-pressing the specific preset button

You can find the link to the firmware here on the Discord forum: 
https://discord.com/channels/1177970799697727548/1177970799697727551/1351951530064089149

onatskiy-sergey, 2025-03-26:
Unfortunately I don't have access, it says not available in discord. Well, I'll wait, I hope it will appear here soon too.

AxxAxx, 2025-03-26:
Hi! 
Now the latest firmware is availible in releases!

closed 2025-03-26 by AxxAxx