AxxAxx/AxxSolder #112: 48V power input

doxygen-spammer, 2025-03-27:
Is it possible to accept up to 60V input by changing just a few parts (buck controller, capacitors, diodes, power switch)?

If so, should the firmware and voltage divider be adjusted, so they can measure up to 60V?

But most interesting: can the power switch and the firmware handle the required lower duty cycles?

Context: I would like to upgrade the JBC CD-2BQF soldering station with an AxxSolder.
The CD-2BQF brings a very sturdy case (heat resistant, drop resistant, ...), but the original firmware has a severe issue: sometimes the station freezes with the heating element turned on, which seems quite dangerous.
Another issue is that the CD-2BQF is very demanding on the AC supply.
While the public power grid works fine, solar inverters cause frequent freezes and error messages.

The 60V input would allow to use a 48V solar battery as power supply.
(End-of-charge voltage is 58.8V.)
The current 9–24V input only allows for 12V solar batteries or 18V power tool batteries, which isn’t that bad already. :)

chillymattster, 2025-04-07:
Hi there,
I really like this project and have been monitoring it for a while now. Great work!

I like the proposal not just because it offers mire flexibility in terms of power supply, it also would enable the use of T470 handle. In my opinion that would boost AxxSolder to the ultimate soldering station replacement by offering the greatest possible freedom in terms of choosing the handle and cartridges. Especially, as T470 handle can take C470 as well as C245 cartridges. 

Cheers

AxxAxx, 2025-06-09:
I will see what I can allow in the next hardware. I think 60V is rather high and would rexuire more expensive and larger components for only a few users who need it.